technyon/nuki_hub #678: Long Delay issue with nuki door lock

bonderaustria, 2025-07-21:
### PROBLEM DESCRIPTION
First of all thank you for your project :) 

Sometimes it takes 7-10 seconds for the lock/unlock command to be sent to the Nuki. What I notice, however, is that I see a retry in MQTT. What I can rule out is the Bluetooth and Wi-Fi reception of the device or the Nuki Lock, which are 1.5 meters away ;)

### REQUESTED INFORMATION
System Information
------------ NUKI HUB ------------
Device: ESP32-S3 (Octal PSRAM)
Version: 9.11
Build: 15758496415.13.1
Build type: Release
Build date: 2025-06-19
Updater version: 
Updater build: 
Updater build date: 
Uptime (min): 201
Config version: 911
Last restart reason FW: RequestedViaWebServer
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
Free internal heap: 181783
Total internal heap: 392543
PSRAM Available: Yes
Free usable PSRAM: 8341048
Total usable PSRAM: 8388608
Total PSRAM: 8388608
Total free heap: 8472868
Network task stack high watermark: 8160
Nuki task stack high watermark: 4848

------------ SPIFFS ------------
SPIFFS Total Bytes: 233681
SPIFFS Used Bytes: 1506
SPIFFS Free Bytes: 232175

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: No
Latest version: 9.11
Allow update from MQTT: No
Update Nuki Hub and Nuki devices time using NTP: No
Web configurator username: ***
Web configurator password: ***
Web configurator bypass for proxy IP: Not set
Web configurator authentication: Basic
Session validity (in seconds): 3600
Session validity remember (in hours): 720
Duo Push MFA enabled: No
TOTP MFA enabled: No
Web configurator enabled: Yes
HTTP SSL: Disabled
Nuki Hub FQDN for HTTP redirect: Not set
Advanced menu enabled: No
Publish free heap over MQTT: No
Nuki connect debug logging enabled: No
Nuki communication debug logging enabled: No
Nuki readable data debug logging enabled: No
Nuki hex data debug logging enabled: No
Nuki command debug logging enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.0.140
SSID: MartinRouterKing
BSSID of AP: 40:ED:00:A1:EA:2A
ESP32 MAC address: 94:A9:90:30:87:44

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: Nuki ESP Hub
DHCP enabled: Yes
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: Yes
Restart ESP32 on network disconnect enabled: No
Disable Network if not connected within 60s: No
MQTT Timeout until restart (s): 60

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: 192.168.0.126
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: nukihub
MQTT SSL: Disabled

------------ BLUETOOTH ------------
Bluetooth connection mode: New
Bluetooth TX power (dB): 12
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10
Enable authorization control: No
Publish authorization topic per entry: No
Max authorization entries to retrieve: 5

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: nukiesphub/
Nuki Hub configuration URL for HA: http://192.168.0.140
Nuki Hub ID: 75347426060692

------------ NUKI LOCK ------------
Lock enabled: Yes
Lock Ultra/Go/5th gen enabled: No
Paired: Yes
Nuki Hub device ID: 1607010367
Nuki device ID: ***
Firmware version: 3.10.7
Hardware version: 4.10
Valid PIN set: Yes
Has door sensor: No
Has keypad: No
Timecontrol highest entries count: 0
Authorizations highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No
Force Lock ID: No
Force Lock Keypad: No
Force Lock Doorsensor: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Auto Unlatch: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Single Lock: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Unlocked Position Offset Degrees: Allowed
Locked Position Offset Degrees: Allowed
Single Locked Position Offset Degrees: Allowed
Unlocked To Locked Transition Offset Degrees: Allowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Unlatch duration: Allowed
Auto lock timeout: Allowed
Auto unlock disabled: Allowed
Nightmode enabled: Allowed
Nightmode start time: Allowed
Nightmode end time: Allowed
Nightmode auto lock enabled: Allowed
Nightmode auto unlock disabled: Allowed
Nightmode immediate lock on start: Allowed
Auto lock enabled: Allowed
Immediate auto lock enabled: Allowed
Auto update enabled: Allowed
Reboot Nuki: Allowed
Motor speed: Allowed
Enable slow speed during nightmode: Allowed

------------ NUKI OPENER ------------
Opener enabled: No

------------ GPIO ------------

Retain Input GPIO MQTT state: No
### TO REPRODUCE
Nothing to do...just lock/unlock the door sometimes 

### EXPECTED BEHAVIOUR
Faster reaction...like the other trys ;)

iranl, 2025-07-21:
Duplicate of #668.

Try the latest build from #675

closed 2025-07-21 by iranl

bonderaustria, 2025-07-22:
there is no info how to install this build..?
is it the development firmware?

technyon, 2025-07-23:
Yes this is development firmware. Download the binary for your ESP here:

https://github.com/technyon/nuki_hub/actions/runs/16473597751

Then flash it via the web interface by selecting the binary, or follow the instruction to do a manual flash in the included text file.