technyon/nuki_hub #680: WT32-ETH01 Ethernet Connection Credentials page could not be opened

drstil, 2025-07-25:
### PROBLEM DESCRIPTION
I use Nuki Hub 9.11 with WT32 ETH-01, connection via Ethernet is set. If I open page Credentials http:/ip</get?page=cred, page stocks and I the the following in Serial Logs. 
```
ETH Disconnected
ETH Stopped
```
Then device is going into reboot fallback, which is also failing because of disabled Wi-Fi fallback. See my logs.
I have to power-off the device, to get back to normal state again.

If I set `Network hardware` to Wi-Fi, I could open the page Credentials without issues!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [ ] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [ ] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [ ] Device used (e.g., ESP32-S3): WT32-ETH01
- [ ] Nuki Hub binary firmware version number used: 9.11
  - [ ] Pre-compiled YES!
  - [ ] Self-compiled
- [ ] Provide the output of http://nukihubIP/get?page=info:
```
System Information
------------ NUKI HUB ------------
Device: ESP32
Version: 9.11
Build: 15758496415.13.1
Build type: Release
Build date: 2025-06-19
Updater version: 
Updater build: 
Updater build date: 
Uptime (min): 1
Config version: 911
Last restart reason FW: NotApplicable
Last restart reason ESP: ESP_RST_PANIC: Software reset due to exception/panic.
Free internal heap: 89924
Total internal heap: 266324
PSRAM Available: No
Network task stack high watermark: 8548
Nuki task stack high watermark: 5116

------------ SPIFFS ------------
SPIFFS Total Bytes: 233681
SPIFFS Used Bytes: 52459
SPIFFS Free Bytes: 181222

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.11
Allow update from MQTT: No
Update Nuki Hub and Nuki devices time using NTP: No
Web configurator username: Not set
Web configurator password: Not set
Web configurator bypass for proxy IP: Not set
Web configurator authentication: Basic
Session validity (in seconds): 3600
Session validity remember (in hours): 720
Duo Push MFA enabled: No
TOTP MFA enabled: No
Web configurator enabled: Yes
HTTP SSL: Disabled
Advanced menu enabled: No
Publish free heap over MQTT: No
Nuki connect debug logging enabled: No
[... 336 lines cut ...]
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4616
load:0x40078000,len:15400
load:0x40080400,len:4
load:0x40080404,len:3176
entry 0x400805a0
Last config version: 911
Current config version: 911
Listing directory: /
  FILE: coredump.hex    SIZE: 44059
Partition size: 2490368
Partition subtype: 16
Nuki Hub version 9.11
Nuki Hub build 15758496415.13.1
Load GPIO configuration
IP configuration: DHCP
Hardware detect: 5
Failed to connect to network. Wi-Fi fallback is disabled, rebooting.

**(Please, remember to close the issue when the problem has been addressed)**

iranl, 2025-07-25:
Probably caused by analogRead(0) while GPIO 0 is used by ETH. Please try 9.12-master4

drstil, 2025-07-28:
@iranl Thank you very much for that quick reply and fix! It solved the issue! Thank you very much! 😆