technyon/nuki_hub #682: GPIO Input unlatch -> MQTT topic always 0

mrbluebrett, 2025-08-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have configured a GPIO with the option Input: Unlatch. If I connect the GPIO to ground, it opens the door as expected. But the MQTT topic sent in nukihub/gpio/pin_13/state is always 0. I have this topic stored in homeassistant as a sensor and it never changes...
I would love to receive a different value when it's not triggered. For example send a 1 always, and a 0 when it detects pulled to ground. After the pulled to ground detection ends, send a 1 again. Or the inverse: 0 always and 1 at the detection, 0 again after.
Then I can know when the unlatch has been triggered in my HomeAssistant because of the GPIO unlatch, and not other reason.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/technyon/nuki_hub/blob/master/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/technyon/nuki_hub/blob/master/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/technyon/nuki_hub/issues)
- [X] Searched the problem in [discussions](https://github.com/technyon/nuki_hub/discussions)
- [X] Searched the problem in the [docs](https://github.com/technyon/nuki_hub)
- [ ] Searched the problem in the [chat](https://discord.gg/9nPq85bP4p)
- [X] Device used (e.g., ESP32-S3): ESP32 devkit 1
- [X] Nuki Hub binary firmware version number used: 9.11
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Provide the output of http://nukihubIP/get?page=info:
```
System Information
------------ NUKI HUB ------------
Device: ESP32
Version: 9.11
Build: 15758496415.13.1
Build type: Release
Build date: 2025-06-19
Updater version: 
Updater build: 
Updater build date: 
Uptime (min): 14
Config version: 911
Last restart reason FW: GpioConfigurationUpdated
Last restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
Free internal heap: 74872
Total internal heap: 266324
PSRAM Available: No
Network task stack high watermark: 8532
Nuki task stack high watermark: 5068

------------ SPIFFS ------------
SPIFFS Total Bytes: 233681
SPIFFS Used Bytes: 50953
SPIFFS Free Bytes: 182728

------------ GENERAL SETTINGS ------------
Network task stack size: 12288
Nuki task stack size: 8192
Check for updates: Yes
Latest version: 9.11
Allow update from MQTT: No
Update Nuki Hub and Nuki devices time using NTP: No
Web configurator username: Not set
Web configurator password: Not set
Web configurator bypass for proxy IP: Not set
Web configurator authentication: Basic
Session validity (in seconds): 3600
Session validity remember (in hours): 720
Duo Push MFA enabled: No
TOTP MFA enabled: No
Web configurator enabled: Yes
HTTP SSL: Disabled
Advanced menu enabled: No
Publish free heap over MQTT: No
Nuki connect debug logging enabled: No
Nuki communication debug logging enabled: No
Nuki readable data debug logging enabled: No
Nuki hex data debug logging enabled: No
Nuki command debug logging enabled: No
MQTT log enabled: No
Webserial enabled: No
Bootloop protection enabled: No

------------ NETWORK ------------
Network device: Built-in Wi-Fi
Network connected: Yes
IP Address: 192.168.XX.XX
SSID: xXXXXx
BSSID of AP: B8:69:F4:XX:XX:XX
ESP32 MAC address: 08:D1:F9:XX:XX:XX

------------ NETWORK SETTINGS ------------
Nuki Hub hostname: nukihub
DHCP enabled: Yes
RSSI Publish interval (s): 60
Find WiFi AP with strongest signal: Yes
Restart ESP32 on network disconnect enabled: No
Disable Network if not connected within 60s: No
MQTT Timeout until restart (s): 60

------------ MQTT ------------
MQTT connected: Yes
MQTT broker address: 192.168.XX.XX
MQTT broker port: 1883
MQTT username: ***
MQTT password: ***
MQTT base topic: nukihub
MQTT SSL: Disabled

------------ BLUETOOTH ------------
Bluetooth connection mode: New
Bluetooth TX power (dB): 9
Bluetooth command nr of retries: 3
Bluetooth command retry delay (ms): 100
Seconds until reboot when no BLE beacons received: 60

------------ QUERY / PUBLISH SETTINGS ------------
Lock/Opener state query interval (s): 1800
Publish Nuki device authorization log: No
Max authorization log entries to retrieve: 5
Battery state query interval (s): 1800
Most non-JSON MQTT topics disabled: No
Publish Nuki device config: Yes
Config query interval (s): 3600
Publish Keypad info: No
Keypad query interval (s): 1800
Enable Keypad control: Yes
Publish Keypad topic per entry: No
Publish Keypad codes: No
Allow checking Keypad codes: No
Max keypad entries to retrieve: 10
Publish timecontrol info: No
Keypad query interval (s): 1800
Enable timecontrol control: No
Publish timecontrol topic per entry: No
Max timecontrol entries to retrieve: 10
Enable authorization control: No
Publish authorization topic per entry: No
Max authorization entries to retrieve: 5

------------ HOME ASSISTANT ------------
Home Assistant auto discovery enabled: Yes
Home Assistant auto discovery topic: homeassistant/
Nuki Hub configuration URL for HA: http://192.168.XX.XX
Nuki Hub ID: 25152445XXXXXX

------------ NUKI LOCK ------------
Lock enabled: Yes
Lock Ultra/Go/5th gen enabled: No
Paired: Yes
Nuki Hub device ID: 129327XXXX
Nuki device ID: ***
Firmware version: 2.15.3
Hardware version: 11.1
Valid PIN set: Yes
Has door sensor: Yes
Has keypad: No
Timecontrol highest entries count: 0
Authorizations highest entries count: 0
Register as: Bridge

------------ HYBRID MODE ------------
Hybrid mode enabled: No
Force Lock ID: No
Force Lock Keypad: No
Force Lock Doorsensor: No

------------ NUKI LOCK ACL ------------
Lock: Allowed
Unlock: Allowed
Unlatch: Allowed
Lock N Go: Allowed
Lock N Go Unlatch: Allowed
Full Lock: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed

------------ NUKI LOCK CONFIG ACL ------------
Name: Allowed
Latitude: Allowed
Longitude: Allowed
Auto Unlatch: Allowed
Pairing enabled: Allowed
Button enabled: Allowed
LED flash enabled: Allowed
LED brightness: Allowed
Timezone offset: Allowed
DST mode: Allowed
Fob Action 1: Allowed
Fob Action 2: Allowed
Fob Action 3: Allowed
Single Lock: Allowed
Advertising Mode: Allowed
Timezone ID: Allowed
Unlocked Position Offset Degrees: Allowed
Locked Position Offset Degrees: Allowed
Single Locked Position Offset Degrees: Allowed
Unlocked To Locked Transition Offset Degrees: Allowed
Lock n Go timeout: Allowed
Single button press action: Allowed
Double button press action: Allowed
Detached cylinder: Allowed
Battery type: Allowed
Automatic battery type detection: Allowed
Unlatch duration: Allowed
Auto lock timeout: Allowed
Auto unlock disabled: Allowed
Nightmode enabled: Allowed
Nightmode start time: Allowed
Nightmode end time: Allowed
Nightmode auto lock enabled: Allowed
Nightmode auto unlock disabled: Allowed
Nightmode immediate lock on start: Allowed
Auto lock enabled: Allowed
Immediate auto lock enabled: Allowed
Auto update enabled: Allowed
Reboot Nuki: Allowed
Motor speed: Allowed
Enable slow speed during nightmode: Allowed

------------ NUKI OPENER ------------
Opener enabled: No

------------ GPIO ------------

Retain Input GPIO MQTT state: No
GPIO 13: Input: Unlatch
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Set one gpio to Input: Unlatch and then connect it to ground for a second. See the mqtt topics sent.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I would love to receive a different value when it's not triggered. For example send a 1 always, and a 0 when it detects pulled to ground. After the pulled to ground detection ends, send a 1 again. Or the inverse: 0 always and 1 at the detection, 0 again after.
Then I can know when the unlatch has been triggered in my HomeAssistant because of the GPIO unlatch, and not other reason.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

technyon, 2025-08-02:
Hi,

I see your use case, but I'd need to think about wether change the "pin_xx/state" is the best way to communicate this. "pin_xx/state" is used for output pins, so it would change it's meaning depending on the configuration.

mrbluebrett, 2025-08-02:
Just for your information and testing, I changed the  GPIO to General Input Pull-Up, and obviously doesn’t open the door, but the MQTT topics work as I expected and requested.

[image]

So you already have this implemented for input pins…

labmaster, 2025-08-11:
> The MQTT is still not working as expected.

But it works exactly as designed ?
If you configure to have a pin set as GPIO (General Purpose Input/Output ) and not assigned to a NUKi funktion then you will get it as GPIO funktion on MQTT.

Its for "General Purpose" .

I see no Problem.

technyon, 2025-08-11:
At this point it works as designed. If you configure it for a specific purpose, it's not general purpose and as such will not trigger the MQTT topic to change. From an API perspective, it would be a lot cleaner to have this in the "trigger" node, but that is driven by the lock/opener and I don't want to start keeping states on both ends.

mrbluebrett, 2025-08-12:
The general input, i don't know why, resets the fingerprint completely...
As a workaround I set an automation in HA to change the MQTT topic to 1 every time it changes to 0 for 1s. For now it's working.

technyon, 2025-08-12:
What do you mean, it resets the fingerprint?

mrbluebrett, 2025-08-12:
I have a fingerprint module with a relay on it. Nothing smart, just a fingerprint reader with a controller. On fingerprint matched, it triggers the relay for 1s. On the relay I have connected the ground on the COM and the NO to the GPIO13 on nukihub. For unlatch it works fine.
But the day I configured the gpio as general input for test, it resetted the fingerprunt module completily to factory. I don't know if general input is putting 3.3v in the pin and when the relay triggers for some ms the module receives 3.3v in the ground, or what is going on there, but it's resetting the control module. This does not happen in unlatch mode.

mrbluebrett, 2025-08-12:
I don’t think my wiring is wrong. It’s quite simple:

5v power supply:
- positive: goes to nukihub-esp32 5v pin and also to dc in + in the fingerprint module.
- negative goes to nukihub-esp32 gnd, to dc in - and also to COM in the fingerprint module.

- fingerprint module NO connected to gpio13 in nukihub-esp32.

I connected as general input pull-up, but I also tried general input pull-down. Can it be I messed up when setting pull down?

mrbluebrett, 2025-08-13:
Still receiven only 0 every unlatch. I keep the connection to ground in the gpio for 1 second.